kauailabs/navxmxp #3: Gyro/Raw Data Update packet too small

BakaBox, 2015-08-16:
The Gyro/Raw Data Update packet is too small for the data.
[IMUProtocol.h - Line 78](https://github.com/kauailabs/navxmxp/blob/master/stm32/navx-mxp/IMUProtocol.h#L78)
```
#define GYRO_UPDATE_TEMP_VALUE_INDEX                38
#define GYRO_UPDATE_CHECKSUM_INDEX                  42
```
The size of ```GYRO_UPDATE_TEMP_VALUE_INDEX``` is 4 when it should be 7 as it uses ```encodeProtocolFloat``` to encode/decode the data

kauailabs, 2015-08-16:
Thanks for reporting this, James, nice find - it's gone undetected for several years and surely resulted in bogus temp readings.  For backwards compatibility reasons changing the packet size will introduce some issues, so we will change this instead to use encode/decodeProtocolUint16, followed by divide by 1000, to get a temp range up to 65C, well beyond the maximum.  I'll send out an update in a few days.

kauailabs, 2015-08-21:
This issue is now resolved with recent check ins.  A full build is available at navx-mxp.kauailabs.com.

closed 2015-08-21 by kauailabs