kauailabs/navxmxp #7: Would like to separate the WPILib specific code in AHRS C++ implementation

virtuald, 2015-12-21:
My intent is to wrap the C++ class with cython so we can use it with RobotPy. RobotPy has it's own version of WPILib, so using the C++ WPILib would most likely cause conflicts.

At the moment, I think the best approach would be to turn the existing code into `AHRSBase`, and remove the `WPILib.h` include and the inheritance from SensorBase et al. The `RegisterIO` interface might be a good boundary here, I haven't looked too carefully at it.

Then, a separate class would be called AHRS, which would have the WPILib specific pieces in it. This would retain backwards compatibility, and be good separation of concerns anyways.

kauailabs, 2015-12-21:
Before embarking on this, there are a couple of design considerations to work out.

- The C++ RoboRIO Library for navX-MXP supports four interfaces:  TTL UART, USB, I2C and SPI.  Thus, the AHRS class uses the WPI library for Serial IO (TTL UART, USB) and Register IO (I2C and SPI)

- The AHRS class inherits from SensorBase, LiveWindowSendable and PIDSource, in order to integrate well with the DriveStation and the PIDController infrastructure.

So abstracting out the Register/Serial IO as well as the PIDController and LiveWindowSendable portions out of the AHRS class is a significant amount of work.  Not impossible, but not trivial either.

In the same way that the C++ and the Java classes are similar but different, my intuition is that a Python library should be different too.

Also, the design should take into account both Serial interfaces, not only because there are many teams using TTL UART and USB, but also because the USB interface is featured on the new navX-Micro sensor, which is compatible w/the navX-MXP interface.

I'm happy to help and work with you, and would like to see a python library for FRC, so let's continue the discussion and work out the best path forward.  Two things that would help me:

- Please post some information about the RobotPy WPILib so that we could analyze the similarities and differences between that and the C++ WPI Library.
- Please also post links to info on how to get started w/RobotPy - I need to understand who I might actually test what might be developed.

virtuald, 2015-12-21:
Hm.

RobotPy's WPILib is pure python, and talks to the hardware via the HAL layer using ctypes, and can be found [here](https://github.com/robotpy/robotpy-wpilib). It does not have a WPILib serial port implementation yet, and the SPI/I2C implementations are very lightly tested at the moment.

The primary reason I was thinking about co-opting the C++ implementation was because I thought I had seen some math-intensive code somewhere -- and given that it's an interpreted language, python isn't particularly good at that sort of thing so it would be better to try and wrap it with cython. However, I looked around a bit, and couldn't find the mathy bits I thought I had seen previously. If that isn't the case, then maybe it'll just be easier to translate it all into pure python and be done with it.

Someone had put together a [hacky implementation](https://github.com/DenfeldRobotics4009/2015_Lopez_Jr/blob/master/imu_simple.py) that uses a serial port, but I'd like to do a little bit better than that.

Information about running RobotPy can be found on our [documentation site](http://robotpy.readthedocs.org), and there are a number of [examples ported from WPILib](https://github.com/robotpy/robotpy-wpilib/tree/master/examples) on the github repo. The current release does not work on the 2016 image yet.

kauailabs, 2015-12-22:
Based on how "fresh" the RobotPy HAL sounds, I'd recommend using the SPI interface.  I2C can get rather complex - more so than SPI - and SPI is about 20X faster (2Mhz for SPI vs. 100Khz for I2C)  - not to mention SPI is pretty easy to debug if we have to break out a logic analyzer.

A port of C++ (or perhaps Java?) to Python is what I'd recommend.  As far as the math operations you mentioned, w/the new v2.0 navX-MXP/navX-Micro firmware that's been released, all of the math intensive operations have been moved down into microcontroller, which is a very capable 100Mhz 32-Bit ARM processor.  That makes the library porting simpler, and uses less cycles on the host robot controller.

There's a bit of thread synchronization code and also time delay code that will need some trivial porting.

So I think that's the best path.

The sources you'll want to look at are four files:
- the AHRS class (for the interface definition, mostly)
- RegisterIOSPI.cpp (here's where you'd map things to the Python HAL SPI IO functions)
- RegisterIO.cpp (for the register value decoding), and
- IMURegisters.h (for the decode helper functions and CRC calculations)  

I quickly reviewed the RobotPy WPILib, it seems at the interface level like identical to the C++ SPI class interface - but of course the devil's in the details.

Once you've got something you'd like me to review, let me know.  I'm interested in getting a RobotPy infrastructure setup here in our lab, let me start spending a few cycles looking into that....

virtuald, 2015-12-22:
Yes, we're trying to keep the python WPILib as close to the Java WPILib as possible, so it's pretty similar. :)

Since all the math stuff is down in the firmware, then a pure python port does seem like the way to go. I'm working on the WPILib 2016 migration for RobotPy first for a few days, then I'll move to the NavX stuff after that I think... we've found that translating Java APIs tends to make more sense. I've been looking forward to playing with the NavX for a few weeks now!

The current RobotPy should be able to co-locate with either C++ or Java programs on the same RoboRIO as long as you haven't filled up your disk. The 2015 releases will only work on a 2015 image of the RoboRIO, we haven't pushed builds for 2016's image yet.

ThadHouse, 2015-12-22:
On the note of porting, do you guys know if there are going to be any major code base changes for the NavX between now and kickoff? I also ported the NavX stuff to the C# WPILib I've been writing, and hope there are no major changes that would require another rewrite.

Also Dustin I have a NavX on my test bench as well, so if you need any help testing the NavX, or any of the 2016 code changes I can help, as mine has the 2016 image too. I can also test any of the Java or C++ changes with the 2016 image if you guys at Kauailabs would like.

kauailabs, 2015-12-22:
Hi Thad,

The navX firmware for the 2016 Build Season is now released, as are the libraries (announcement just went out on ChiefDelphi).  We just released the 2.3 firmware which has some minor fixes related to the TTL UART/USB interfaces.  That's recommended for your testing.  The C++/Java libraries and samples have been unchanged for several months now; the recent activity just completed was the work w/Team 900 (Zebracorns) on their new V2 LabView Library.

Looking back over the past few months, the libraries did undergo a rewrite about 3-4 months ago and new samples were added.  Functionally, a few functions were moved down to the microcontroller (the zeroing of the yaw, and the velocity and displacement integration [that last bit is accurate to 1 meter every 15 seconds, so I'm not trying to get anyone's hopes up as to that it can be used for autonomous displacement tracking]).  But if you forked this code within the last few months, you should be fine.

As to testing, we've been doing a lot of it here lately, but I will absolutely never turn down any offer to help test.   And I would be happy to try out the C# library.  I'd like to work out some arrangement whereby KauaiLabs can provide a link and some info about these new Python and C# developments that are underway and/or complete....

ThadHouse, 2015-12-22:
I forked it about a week ago, and looking at the commits since then it doesn't look like anything has changed. I was a little disappointed that the FPGA MXP issue still hadn't gotten fixed, but that seemed like a FIRST issue, and knowing a simple workaround makes it not a problem for us.

The C# libraries can be found here
https://github.com/robotdotnet/WPILib

The NavX stuff is in the Extra project. Note everything has been upgraded to 2016 spec, so you would need a 2016 image to test.

kauailabs, 2015-12-22:
I don't have a 2016 image yet (didn't make it onto the beta team, unfortunately) - that's the one area where the libraries might change a small bit once we finally do get that and the beta.

Took a quick look at the C# libraries, I'm guessing you used the Java  lib as a starting point.  I noticed that you had the most recent change (which was a reversing of the pitch/roll register values), so I'm confident you're in synch there.

Are you still seeing the FPGA MXP issue?  I have never got definitive word from NI as to whether they found anything.  As noted in a conversation w/you earlier, we did discover that there were cases where an internal I2C bus glitch could cause the navX-MXP to stop sending data out the communication interfaces, and the fix for that went in around the 2.1 timeframe.  We've tested extensively that there are no issues in this area, but would very much appreciate it if you could go over that area in your testing.

ThadHouse, 2015-12-22:
I will do a lot more testing in the next few weeks to see if I can reproduce it again. I thought I saw it when I was porting the code, but I think at the time I was testing with some wrong I2C code. I'll update to the newest firmware and do more testing over the next few weeks. Hopefully its gone, and it was just my testing code that made me think I saw it.

It looks like 330 submitted a pull request to you guys for the 2016 image, and the things changed there are the only things I remember changing anyway.

And yeah it was ported from Java, as it only took about 20 minutes to copy paste, and refactor the few differences needed to be made. That's why all the documentation is messed up. I'm working on fixing that today.

kauailabs, 2015-12-22:
Are you using I2C?  I'd highly recommend SPI, it's much faster than I2C and if you're using the C++ or Java WPI Lib (which don't support large burst reads on I2C) I2C is way slower than SPI.  I am very happy w/the SPI performance and reliability, for C++ and Java I2C is just not where I want it to be - I'm hopeful that this has been resolved in the latest WPI libraries.  There's some more discussion on that on the new wiki (http://navx-mxp.kauailabs.com/guidance/selecting-an-interface/).  All of our samples are now using SPI, and that's what we're recommending to anyone who asks.

The pull request from Joe Ross was just the PIDSourceType change, I'm guessing you hit that while porting.

Do you mind if we put a link to your C# work online once you're ready?  I realize it doesn't make sense until the 2016 firmware/libraries are available, but I'm thinking if it was available at kickoff, it could help some others.

ThadHouse, 2015-12-22:
We used SPI last season and plan on using it again. We saw the issue last year on both SPI and I2C however. Maybe it was just an issue with the LabVIEW libraries. This year we are switching. I don't have access to the 2016 LabVIEW beta, so I won't be able to test them there. But we would be using SPI this year as well, and since we know the workaround we are fine using it.

Yeah, the PIDSourceType was the only change that affected the code I believe. The I2C interface has not changed at all, and its still limited to 8 bytes at a time from the looks of things. 

We would be happy for you to put a link to the C# work. Because of the rules, I have to release the code before Kickoff, and I was going to do that probably the Thursday or Friday before kickoff. Its pretty much ready for release now, just want to verify it with the final image when we get it. I should be able to port the examples by then too, so if you want a copy of those I should be able to get them to you.

sciencewhiz, 2015-12-23:
The I2C 8 byte restriction was fixed this summer. https://usfirst.collab.net/gerrit/gitweb?p=allwpilib.git;a=commit;h=c21c47f546e5f30b49cad58634591c86399c07fe

I did some testing with the NavX, but not enough that I felt comfortable taking out the 8 byte workaround with my pull request.

kauailabs, 2015-12-23:
Good to hear, sciencewhiz, I'm looking forward to getting ahold of this WPILib update, because it's going to mean that I2C is much more usable for data-intensive sensors like navX-MXP.  I still think SPI's a superior interface for navX-MXP (due to the higher clockrate) but navX-MXP (and navX-Micro) definitely get better performance for I2C once this is publicly available.

ThadHouse, 2015-12-23:
Interesting. I had forgotten about that. I changed MAX_WPILIB_I2C_READ_BYTES to 100,000 on my copy of the code, and it seems to be  working fine. I'll leave it running for a few hours and make sure it stays running and report back. Is that the only number that should need to be changed?

virtuald, 2015-12-23:
@kauailabs if you send an email to Brad Miller, you might be able to get access to the Beta test image. They gave Thad and I access, and neither of us are on beta teams -- but we're doing things for the greater good.

kauailabs, 2015-12-23:
Thanks @virtuald, I appreciate the advice, I'll give that a try.

@ThadHouse, from the navX-MXP perspective, if we can get a contiguous read of 111 (0x6F) bytes, we're good.   I think a logical maximum is 256 bytes (given the 8-bit I2C addressing).  It's interesting (perhaps) to note that sensors w/more than 256 bytes of register space (e.g., the Bosch BNO055) use a bank-switching technique when accessing registers, so I believe this is a safe maximum given the I2C protocol specification and current practice by well-known silicon vendors.

ThadHouse, 2015-12-23:
@kauailabs Right now its telling me its sending and reading a single 108 byte packet. If I set MAX_WPILIB_I2C_READ_BYTES back to 7, it sends multiple 7 byte packets, and then a 3 byte packet. So it definitely seems like large byte array readings are working. And its been stable for the past 20 minutes.

kauailabs, 2015-12-23:
@ThadHouse That's great news.  What time is it in Oregon, must be about midnight? Anyways, hopefully I'll be able to replicate this myself soon.

Please note that the navX-MXP SPI protocol inclues a CRC, whereas the I2C protocol does not.  It was added to SPI protocol because we envisioned supporting the higher bitrates with it and were concerned about detecting data corruption if noise was encountered.   I *wish* we had done this for I2C (it's a breaking change to fix it now), but it's also less of an issue because of the lower I2C bitrate and it's less "natural" to do so, because I2C is really designed to be a "read from registers" protocol.   Just wanted to be clear that because of this, even if we get a single contiguous burst read, that SPI will still be theoretically superior both because of higher bitrate as well as because of the protocol error detection.   The RoboRIO libraries will discard any SPI read where there's a CRC failure, but will have invalid data w/an I2C read.  I'll take the blame for the design issue.  Just wanted to make sure you were aware of it.

ThadHouse, 2015-12-23:
It's just approaching 11 now, so its not too late.

We still plan on using SPI anyway, but having I2C even a little faster could be very helpful.

sciencewhiz, 2015-12-23:
When I looked at what it would take to update for the beta 2016 C++ changes, it was more then just the PIDSourceType, because the Task class changed significantly. I played with it a little and couldn't get the task part working and gave up.

kauailabs, 2015-12-23:
Thanks, @sciencewhiz.  I just received access to the 2016 WPI library beta and will start looking into it.

virtuald, 2016-01-02:
@kauailabs I think I'm mostly through my 2016 updates for RobotPy, so I'll most likely take a shot at porting this during the weekend. I'll let you know how it goes.

virtuald, 2016-01-07:
I'm going to close this, as it's not necessary. I've got a pure python port working most of the way, just have some offsets incorrect or something. Tracking my progress at https://github.com/robotpy/robotpy-wpilib-utilities/issues/9

closed 2016-01-07 by virtuald