kauailabs/navxmxp #13: Provide release artifacts on Maven repo

rytse, 2016-06-22:
Could kauailabs please put the Java Roborio library (navx_frc.jar) on a Maven repo so that users can add  the navx library as a dependency? Thanks.

leoforney, 2017-06-01:
Is any work being done on this?

kauailabs, 2017-06-05:
This item is currently assigned low priority, as we are currently focusing our efforts on other new features and products.

dennisgove, 2017-08-31:
Has there been any additional movement on this?

kauailabs, 2017-09-01:
We've got the latest navX-MXP FRC Java library available on a maven repository now.  We could use some help testing it.  Instructions on how to reference the repository are here:  http://www.pdocs.kauailabs.com/navx-mxp/software/roborio-libraries/java/

See the "Maven Repository" section at the bottom of the page - and please let us know how this works for you.