kauailabs/navxmxp #33: Provide ZIP download of just Java and C++ artifacts

rhauch, 2017-01-02:
The [current download]() includes a `setup.exe` executable that extracts and installs the navX-MXP software and libraries, but this works only on Windows. Users that are on MacOS or Linux do not have a way of accessing the library files for Java and C++.

Please provide a second downloadable ZIP file that includes only the library files and examples for Java and C++ (really, everything already in the `java` and `cpp` directories created in the installation by the current installer) but can be used by anyone running Windows, MacOS, or Linux.

sciencewhiz, 2017-01-02:
It already exists. See bottom of http://www.pdocs.kauailabs.com/navx-mxp/software/

kauailabs, 2017-01-02:
Closing as "already-implemented".

closed 2017-01-02 by kauailabs

rhauch, 2017-01-02:
@sciencewhiz, @kauailabs Great to see! Perhaps it'd be useful to add a link to [the Java page](http://www.pdocs.kauailabs.com/navx-mxp/software/roborio-libraries/).

The existing ZIP is for the official release, and I don't see a [download](http://www.kauailabs.com/public_files/navx-mxp/) for the beta version. Any idea when the beta version will be released and available this way?

kauailabs, 2017-01-02:
Release is planned for this Friday; other cool stuff is coming, so stay tuned to ChiefDelphi or @Kauailabs twitter/Facebook accounts.

I'll look into adding links to the Linux/MacOS libraries for Java and C++ on those respective pages, thanks for the suggestion.

rhauch, 2017-01-02:
Awesome!

rhauch, 2017-01-03:
@kauailabs Any idea why a [build](https://travis-ci.org/strongback/strongback-java/builds/188432227) of mine on Travis-CI might get a 403 Forbidden while trying to get http://www.kauailabs.com/public_files/navx-mxp/navx-mxp-libs.zip when that same build can get to other web resources (like https:github.com and http://first.wpi.edu)? Same build runs fine locally, and my laptop can download that ZIP file just fine. Might some IP addresses under [Amazon's EC2 ranges](http://docs.aws.amazon.com/general/latest/gr/aws-ip-ranges.html) (which travis-ci.com uses) be blocked by kauailabs.com?