kauailabs/navxmxp #52: VISA Framing Error

steven-campbell, 2017-02-20:
While viewing my DS logs from driver practice, I noticed the following message. Is this an issue? Is there any solution?

` java.lang.RuntimeException:  Code: -1073807253. HAL - VISA: Framing Error
at com.kauailabs.navx.frc.AHRS$IOThread.run(AHRS.java:1242)
at edu.wpi.first.wpilibj.SerialPort.read(SerialPort.java:256)`

kauailabs, 2017-02-20:
Keep us posted if you see continued occurrences.  In theory the code should be resilient to this case, and output a console message if it does occur.  Sounds like RoboRIO encountered an error.  Perhaps the connection is loose and thus subject to connectivity errors.  Are you running USB?  Did the sensor continue operating as expected during the time this occurred?

steven-campbell, 2017-02-24:
We had some time to test on our practice robot yesterday. The sensor does return expected values even as the error continues to appear. We have the sensor connected via the MXP port on the roboRIO. we also checked the connection and there does not appear to be any issues there. We still would like to have this error not appear at all because it is causing a large number of logs to appear in the DS log viewer.

kauailabs, 2017-02-24:
One option you have is to change the interface you are using to SPI or I2C - both are available if you have a navX-MXP connected to the MXP port; these interface options do not use the serial port and thus should not be susceptible to the framing error you are seeing.  If that doesn't work out, we can arrange a board exchange if you'd like (in which case you'd want to contact support@kauailabs.com).

steven-campbell, 2017-02-27:
This solved it. We switched it to SPI and the error has disappeared! Thank you for all the help!

closed 2017-02-27 by steven-campbell