kauailabs/navxmxp #53: Issue with NAVX Micro Over I2C

bigabos-hey, 2017-02-20:
After upgrading to 2017.3.1, I2C communication with NAVX micro failed.  USB communication still functioning.  Roll back to 2017.2.1 resolved issue.  Reference issue #490 in WPILib repo... https://github.com/wpilibsuite/allwpilib/issues/490

sciencewhiz, 2017-02-20:
I2C return values were corrected in wpilibsuite/allwpilib#484

bigabos-hey, 2017-02-20:
Not sure what you're saying here.  That "fix" was included in 2017.3.1 which broke the communications with the NavX.  Is an updated needed to the NavX firmware?

kauailabs, 2017-02-20:
The navX Java/C++ Libraries use the WPI Library I2C class write() and read() methods, and assume that these methods will return true upon success, false in case of error.  If that is no longer the case, the navX roborio-side libraries will need to change.  Sciencewhiz, can you verify that w/2017.3.1 release that false return from the I2C write/read methods now indicates success?

sciencewhiz, 2017-02-20:
^ @calcmogul

I think the issue that was fixed was it would return fail for many cases when it succeeded.

virtuald, 2017-02-21:
The underlying HAL API definitely works, as RobotPy is able to use it correctly. Of course, we were already doing the error checking correctly anyways so not much changed in 2017.3.1 for us... 👍

kauailabs, 2017-02-21:
@virtuald, @sciencewhiz:  I've installed 2017.3.1 into eclipse, rebuilt and the navX Java Data Monitor works fine for SPI, but I2C is not working now, similar to as-reported by @bigabos-hey.  I don't think the issue is the return code.  What I"m seeing is both the write and the read requests are returning true (which per the documentation means the transfer was aborted - and they take awhile to return so that may be what's actually occurring).  And the read buffer has 0 bytes in it.

I will agree that the navX library treatment of the return result codes from I2C.write and I2C.read is reversed in sense from what is documented.  It's always worked until now.

But it really seems like something else is wrong.  And the fact that true is returned from the read (which docs say means transaction aborted) corroborates this.

No firmware has changed, only the WPI libraries.

Can you try this and let me know your results?  This code's been working for two years now in the navX library - it sure seems like something's now broken in the WPI library.

kauailabs, 2017-03-03:
This has been resolved with release 3.0.329, which is now available for download from the [navX-MXP software page](http://www.pdocs.kauailabs.com/navx-mxp/software/).

What changed:  The return result (a boolean) from WPI Library I2C.write() and I2C.readOnly() methods was previously (before 2017.3.1) returning TRUE for success, but as of 2017.3.1 it is returning FALSE for success.  The RegisterIO_I2C.java and RegisterIO_I2C.cpp files therefore changed to invert the sense of these variables when communicating with navX-MXP.

NOTE:  navX-MXP RoboRIO C++/Java Library version 3.0.329 and above will no longer work correctly with WPI Library versions before 2017.3.1.  The change to the behavior in the WPI Library 2017.3.1 is a *breaking* change, in that the interface to client applications was modified.

closed 2017-03-08 by bigabos-hey