kauailabs/navxmxp #59: Delay in LabView USB communication, I2C Sometimes gives error -1

ranlevinstein, 2017-03-18:
We are programming in Labview and we have some problems using navx this year.
When we connect it in USB, we get delayed angle data.
When we connect it via I2C, it works until we get error -1 in I2C write and read actions.
We made sure that both our navx and labview libraries have been updated to the most recent versions.
Any help will be greatly appreciated.

aeastet, 2017-03-18:
It could be the sample rate. I'm guessing you are sampling to slow or updating too slow. I would be happy to look at your code if you send a link.

aeastet, 2017-03-21:
I looked through your code. It looks like you are using an old version of the navX software. I would recommend downloading the news version of the software.  You can sample as fast as 200Hz with this new library. It is also very easy to integrate. I think this would get rid of the -1 error that you are seeing. I made sure that we recover from errors and even loss of communication with the new library. 

You have some cool stuff in your code. You should look into state machine for autonomous. I am doing most of the same stuff in my code. You guys seem to go a little farther in some places then we do. But overall great job. One other suggestion NI (National Instruments) teaches in their classes that all code should fit into one window. It makes the code much easier to troubleshoot.

aeastet, 2017-03-21:
OK I looked further. I do see the old navX code and the new navX code. Do you get the -1 error on the new code? How long of a USB cable are you using for the navX? Do you have both versions of the navX integrated?

ranlevinstein, 2017-03-21:
We saw the error in the new code, the usb cable is about 1 meter long and we had problems with the navX early in the season so we tried to use it with some old versions but it didn’t work well so we used another gyro for a while but now we wish to use the navX again. We have both the new code for the navX and the old code but we only use the newest version.

aeastet, 2017-03-21:
What sampling rate are you using when you see the -1 warning? I looked up the error code -1 and it does not have a description. Do you get readings from the navX when you are receiving the -1 warning? What USB port are you using on the roborio?

ranlevinstein, 2017-03-21:
We tried 50HZ, 100HZ, 200HZ and all had problems. We get the -1 error only when it’s connected via I2C and we don’t get any new values from the navX when we get the error. We tried Both USB ports when we connected it via USB.

aeastet, 2017-03-21:
Do you have more than on navX? If so do you see the same problem with the second one? Do you have a shorter cable that you could try running with?

ranlevinstein, 2017-03-21:
Actually we tried it with 3 different navX and all had the same problems, we also tried it with 10 cm cable sometime and I remember it still had the problem. We would like to connect the navX via I2C so if there is any way the solve the problem with it, it will be better for us than solving the problem with USB.

aeastet, 2017-03-21:
I have not seen this error so far with development. I will try to put together the I2C connection and see what I can do to replicate this problem. I would expect that it has something to do with the communication. When was the last time that you downloaded the navX software? We has a release just about three weeks ago. I had to add a wait between the write and the read. That may help your problem.

ranlevinstein, 2017-03-21:
Thank you very much for all of your help!
We uninstalled the old software and downloaded and installed the new one in the end of last week.

aeastet, 2017-03-22:
I just want to make sure that you are connecting the navX through this connection on the roborio.

[image]

Please take a look at the picture an confirm that you are using this connection.

Thank you,

Tim

ranlevinstein, 2017-03-22:
This is the connection we used.

aeastet, 2017-03-22:
OK we have done a little digging and found a few interesting things. Most of these affect USB but we are trying to trouble shoot a few things here and this information would be helpful. 
What camera are you using for vision?
What resolution are you running at?
Is the camera plugged into the roborio?

ranlevinstein, 2017-03-22:
When we did our tests the camera wasn’t plugged to the roborio but its still defined in our code. The camera is Microsoft life cam hd 3000, it sent us images with resolution of 160x120 pixels.

aeastet, 2017-03-22:
Have you tried taking the camera code out to see if the error goes away? How hard would it be for you to try and see if you take that code out if that helps with the error that you are seeing? I am trying to eliminate what I can. I am planning to try and really dig into this this weekend when I have some time. I am hoping to have a solution by Monday if I can.

ranlevinstein, 2017-03-22:
We will try when we have the option. We really prefer to connect it via i2c and not usb. In Tuesday we have our last competition so our schedule is a little tight and I hope that we will have time to try to use the navX in Monday but it’s probably too late for us.

ranlevinstein, 2017-04-03:
Any luck with I2C?

aeastet, 2017-04-04:
I haven't had a chance yet. I have a lot going on at the moment. I am going
to try to make time for this very soon.