kauailabs/navxmxp #63: large delay in SPI, and conflicting comment about it

PeterMitrano, 2017-11-26:
https://github.com/kauailabs/navxmxp/blob/341c96f20816cc97a34341798f4f26c4cdb7ca3e/roborio/c%2B%2B/navx_frc_cpp/src/RegisterIOSPI.cpp#L50-L51

My first question is why is there a 1ms delay here? May I remove/shorten this delay or at least update the comment? 1ms is a very long time to be blocking in SPI world.

kauailabs, 2017-11-26:
Since the SPI protocol is implemented on the navX-MXP STM32 F4 microconrtroller in an interrupt service routine, a small delay (worst case of 200 microseconds) is necessary; the only higher priority interrupt on the micro controller is the systick interrupt, and the SPI interrupt will preempt lower priority interrupts.  If a method to achieve a 200 us delay is available on the roborio, the existing delay could be decreased to that.  The practical minimum delay is not fully characterized, which was what the comment was meant to indicate.  At the highest rate (200Hz) the existing implementation is not known does not miss any samples, so it’s not clear what decreasing the delay will achieve although lower latency is one possibility.

PeterMitrano, 2017-11-26:
Ok, seems legit. I'm not actually using SPI I just wanted to bring it up in case it was unnecessary or a bug.

closed 2017-11-26 by PeterMitrano