kauailabs/navxmxp #82: NavX does not report using FRC's usage reporting

sciencewhiz, 2019-10-14:
In the[ usage reporting spreadsheet from 2019](https://www.firstinspires.org/robotics/frc/blog/2020-beta-testing-usage-report), the NavX is not reported.

kauailabs, 2019-11-14:
@sciencewhiz, I've added usage reporting for C++, Java (now available on the navX software download page, as version 3.1.390).

Do you know how one can add Usage Reporting from LabVIEW?  I couldn't figure that part out....

sciencewhiz, 2019-11-16:
You can use this VI Snippet for usage reporting in LabVIEW. 
[image]

kauailabs, 2019-12-13:
@sciencewhiz, we just release 3.1.400 of the navX-MXP RoboRIO software; this includes updated LabVIEW VIs which use the code snippet you've provided to update the usage reporting.  Thanks for the help!

closed 2019-12-13 by kauailabs