kauailabs/navxmxp #88: WPILib simulation support

virtuald, 2020-01-15:
Are there plans to support simulation? It should be pretty straightforward to add a SimDevice.

For example:

* https://github.com/wpilibsuite/allwpilib/blob/master/wpilibc/src/main/native/cpp/ADXL345_SPI.cpp#L22
* https://github.com/wpilibsuite/allwpilib/blob/master/wpilibc/src/main/native/cpp/ADXL345_SPI.cpp#L64

kauailabs, 2020-01-30:
Thanks; we've just released a version with SimDevice support for C++ and Java.

To acquire it, use the online vendordeps URL (https://www.kauailabs.com/dist/frc/2020/navx_frc.json).  Update your VSCode project using this, and it'll use the library version (3.1.403) which also includes a windowsx86-64 platform distribution.

closed 2020-01-30 by kauailabs

virtuald, 2020-01-31:
Looks like it works, thanks! We're actually compiling from the sources zip, which worked great.