kauailabs/navxmxp #94: Publish linuxx86-x64 archives to maven

Modelmat, 2020-02-05:
Running the `./gradlew build` on my local machine produces all of these maven zips:
[image].

This libraries *builds* for linuxx86-x64 fine on Linux.

I was wondering if it would be possible to publish these to maven, as it would allow for the possibility of Linux desktop simulation support (which currently doesn't work).

kauailabs, 2020-02-05:
It's possible, will start looking into what's involved.  Don't have a good way to test it, so will need your help w/that.  Will update on this issue as things progress.

Modelmat, 2020-02-05:
Okay, thanks.
I believe all that's needed is a Linux machine for building and testing.

kauailabs, 2020-02-05:
It's a drag the cross-compiling isn't supported, and it requires setting up a different machine for this to manage the linux simulation build process.

Are you aware of anyone else who might be wanting linux simulation capabilities?  We must trade off the effort to do this versus developing our next-generation sensors.  You're the only one who has asked, so if this is a one-off effort I'm not sure it's worth it from our side.

auscompgeek, 2020-02-14:
Would `"skipInvalidPlatforms": false` do the trick? (Disclaimer: I have little idea how GradleRIO works.)

Modelmat, 2020-02-14:
I shouldn't think so - this issue that there are no platform builds for desktop on Linux.

virtuald, 2020-02-27:
@kauailabs RobotPy compiles navx for Linux/Windows/OSX all using github actions -- I've been pretty happy with it compared to other CI solutions I've used in the past. Since you're completely open source (unlike CTRE, for example), using github actions might make sense for you and you'd be able to support all the simulation platforms, unlike most vendors.

dawonn, 2021-01-10:
Just came to say that I'm interested in this feature as well. (FRC2021 kickoff) 

```
> Task :linkFrcUserProgramLinuxx86-64ReleaseExecutable
/usr/bin/ld: /home/dereck/frc/SimDemo/build/objs/frcUserProgram/linuxx86-64/release/frcUserProgramCpp/cpxox320jmytzosewdh0ktd87/Drivetrain.o: in function `Drivetrain::GetYaw()':
Drivetrain.cpp:(.text+0x10): undefined reference to `AHRS::GetYaw()'
/usr/bin/ld: /home/dereck/frc/SimDemo/build/objs/frcUserProgram/linuxx86-64/release/frcUserProgramCpp/9760xtgdau9jsoxxi90ozrpww/Robot.o: in function `Drivetrain::Drivetrain()':
Robot.cpp:(.text._ZN10DrivetrainC2Ev[_ZN10DrivetrainC5Ev]+0x17a): undefined reference to `AHRS::AHRS(frc::SPI::Port)'
/usr/bin/ld: Robot.cpp:(.text._ZN10DrivetrainC2Ev[_ZN10DrivetrainC5Ev]+0x4dc): undefined reference to `AHRS::ZeroYaw()'
collect2: error: ld returned 1 exit status
```

person4268, 2022-07-14:
I would also like to see this, especially considering that it doesn't seem like you're open sourcing your newer code anymore.