kauailabs/navxmxp #101: WPILib 2022 Deprecations

lhvy, 2021-07-23:
`PIDSource`, `Sendable` and `SendableBase` have been deprecated in WPILib 2022. All three are used by the latest navx_frc build. When building robot code making use of the navx library, the following errors occur:

```
> Task :compileJava
NavXGyroscope.java:156: error: cannot access PIDSource
        return -ahrs.getYaw();
                    ^
  class file for edu.wpi.first.wpilibj.PIDSource not found
NavXGyroscope.java:162: error: cannot access Sendable
        return ahrs.getRoll();
                   ^
  class file for edu.wpi.first.wpilibj.Sendable not found
NavXGyroscope.java:168: error: cannot access SendableBase
        return ahrs.getPitch() - basePitch;
                   ^
  class file for edu.wpi.first.wpilibj.SendableBase not found
```

Here are the relevant PRs in WPILib:

https://github.com/wpilibsuite/allwpilib/pull/3210
https://github.com/wpilibsuite/allwpilib/pull/3236
https://github.com/wpilibsuite/allwpilib/pull/3432

mdurrani808, 2021-12-04:
Any updates on this?

lhvy, 2021-12-04:
I ended up emailing Kauai Labs a while ago, and they do have a beta release which we can use and test. The online install .json url is `https://www.kauailabs.com/dist/frc/2022/navx_frc.json` (This can be used in VS Code with the WPILib extension installed: Click the WPI icon in the corner to open the WPI Command Pallet > `Manage Vendor Libraries` > `Install new library (online)`)

Unfortunately our team only just got physical access to our robot again after lockdown, so we haven't been able to test yet, but the beta release is certainly enough for our code to build and pass unit tests.

sciencewhiz, 2022-02-15:
Is this repo still used? The 2022 build has changes not hosted here

kauailabs, 2022-02-15:
This repo is inactive, since the introduction of navX2-MXP, which is in a private repo that requires a NDA, as we have experienced several attempts to copy or hijack the project.

We leave the original inactive repo available for reference to those interested in learning about navX-MXP - and have made some attempts to keep it in synch with latest wpi lib changes, but it’s extra work that’s not high priority.   So while the firmware is up to date in this repo, the wpi lib bits are likely not.