kauailabs/navxmxp #102: How to setFusedHeading in navx mxp

mendax1234, 2021-09-29:
We use the first version of navx , not the second. How can we setFusedHeading in it?

kauailabs, 2021-09-29:
Do you mean getFusedHeading()?

This is an advanced feature, requiring careful magnetometer calibration and use within an environment without large magnetic anomalies.  Unless the magnetometer has been properly calibrated, and is in an environment without magnetic field disturbances.  This feature is not typically recommended for FRC robotics teams.

If the magnetometer is not calibrated, then the value from getFusedHeading() is identical to the value returned from getYaw().

Also note that even a calibrated magnetometer is less accurate than a gyroscope over a short period of time.  So by using fused heading, you may be introducing unwanted errors.  So again, this is an advanced feature.

Link to [Magnetometer Calibration Info](https://pdocs.kauailabs.com/navx-mxp/guidance/magnetometer-calibration/).

Also, the [navXUI page](https://pdocs.kauailabs.com/navx-mxp/software/navx-mxp-ui/) describes how some of this information is displayed.  Fused Heading is called "Heading" in the navXUI.  If you do choose to use it, you can monitor it's behavior here.

mendax1234, 2021-09-30:
Emmm , thanks . l just want to achieve field-oriented mode on my swerve. l now use getYaw(), however , the field-oriented doesn't seem to work well. l try to look at the Yaw angle when l rotate my robot. When l rotate from 0 to 45 in clockwise direction , the angle seems to be -168.2699. l don't know why it be this ?

kauailabs, 2021-09-30:
>    When l rotate from 0 to 45 in clockwise direction , the angle seems to
be -168.2699.

Which angle?  getYaw()?  Or getFusedHeading()?

Orientation is important.  Is the board mounted following the guidelines
here
<https://pdocs.kauailabs.com/navx-mxp/installation/roborio-installation/>?
One of the board axes needs to be perpendicular to the earth.  And if the
board Z axis is not pointing up, then you'll need to run omnimount to
calibrate the board to use that orientation.

mendax1234, 2021-09-30:
GetYaw

l check it again, it seems that it is installed on the roborio correctly.

>>>One of the board axes needs to be perpendicular to the earth.  And if the
board Z axis is not pointing up, then you'll need to run omnimount to
calibrate the board to use that orientation.

What does this mean? If l install it just like the docs shows , is that correct? Do l need to calibrate again?

mendax1234, 2021-09-30:
Besides, how to set the navx's yaw angle increase for counter clockwise

kauailabs, 2021-10-01:
>  If l install it just like the docs shows , is that correct?

If you install it like shown on the "Roborio installation page", and as
long as the roborio is "flat" (not tilted) with respect to earth, then yes,
by definition the Z axis will be "up", and the Z axis will be perpendicular
<https://en.wikipedia.org/wiki/Perpendicular>to the earth.  This meets the
conditions for correct orientation of the navX-sensor's 3D coordinate
system with respect to the robot.

You can learn more about the 3D coordinate system on the Terminology
<https://pdocs.kauailabs.com/navx-mxp/guidance/terminology/>page.  There is
lots of helpful and educational info in this section and other pages in the
"Guidance <https://pdocs.kauailabs.com/navx-mxp/guidance/>" section.  For
best results, it's recommended to review this material carefully.

> Do l need to calibrate again?

It's recommended.  And it's absolutely required if the last time Factory
Calibration occurred the conditions described above were not met.

After verifying orientation, and after successfully performing Factory
Calibration (reminder:  the board must be still during the Factory
Calibration), then you can use the navXUI.  If you reset the yaw so that
it's zero, you should then be able to rotate the roborio and see the angle
updated in the navXUI.

- scott

kauailabs, 2021-10-01:
>  how to set the navx's yaw angle increase for counter clockwise

Multiplying the returned yaw value by -1 seems like a very simple,
straightforward solution.

mendax1234, 2021-10-01:
Thanks very much， where can navxui be installed？

kauailabs, 2021-10-01:
See instructions at bottom of navXUi page linked to in my last email.

mendax1234, 2021-10-01:
Thanks，l will soon dig into it.

mendax1234, 2021-10-01:
Really Thanks. After testing my navx, l have found where is the error. l instantiate the ahrs in a heading file , and this bring all strange bugs. After instantiating it in the "cc" file, everything goes well. Now the field-oriented is achieved.

Really thanks for you idea to debug. l first use navxUI to ensure that my navx is working well. Them , l dig into the code and find this really small but fatal bug!

closed 2021-10-01 by mendax1234