kauailabs/navxmxp #107: FRC Roborio depricated feature

error2403, 2023-01-09:
The new 2023 wpilib library has removed the RuntimeDetector.isRaspian() method. All of your AHRS constructors utilize this method so there is no way for us to construct an AHRS in our robot code without hitting this error.

ParkerMeyers, 2023-01-10:
I also noticed this, in the meantime is there a workaround?

error2403, 2023-01-10:
I found another library to use: https://dev.studica.com/releases/2023/NavX.json

This seems to work for the testing I did earlier today.

ParkerMeyers, 2023-01-10:
> I found another library to use: https://dev.studica.com/releases/2023/NavX.json
> 
> This seems to work for the testing I did earlier today.

Oh awesome! Thank you so much

danielcello, 2023-01-28:
+1 on this